dusty-nv/ros_deep_learning #1: Package not finding jetson-inference

pmcrivet, 2016-11-04:
I cloned, and make the jetson-inference package on the jetson tx1 and it's working great ( /home/ubuntu/jetson-inference).

But when I try to compile ros_deep_learning, it gives me an error that it cannot find jetson-inference package. Do I have to set some paths in the CmakeList or elsewhere?

Thanks for this awesome DeepLearning demo!

look-recognize, 2016-12-27:
I met the same problem!!

look-recognize, 2016-12-27:
@pmcrivet I think you might had not "make install" the jetson-inference. So the easiest way is rewriting the cmakelists by replacing jetson-inference item with the path of file "jetson-inference.o".

eduagrobot, 2017-01-19:
Hi, could you show me your cmakelist? I don't know how I replace that

Thanks in advanced

nPellejero, 2017-02-10:
Hi, I met the same problem, there is some solution?

dusty-nv, 2017-02-10:
Can  you try appending this to jetson-inference/CMakeLists.txt then running `make install`?

```# install header files
foreach(include ${inferenceIncludes})
    install(FILES "${include}" DESTINATION include/jetson-inference)
endforeach()
# install the shared library
install(TARGETS jetson-inference DESTINATION lib/jetson-inference EXPORT jetson-inferenceConfig)
# install the cmake project, for importing
install(EXPORT jetson-inferenceConfig DESTINATION share/jetson-inference/cmake)```

nPellejero, 2017-02-10:
Yes, just did it and worked!

Thank you dustin !!

dusty-nv, 2017-02-10:
Thanks for confirming.  I've updated the CMakeLists in jetson-inference master, [9f6a238](https://github.com/dusty-nv/jetson-inference/commit/9f6a238ae00f26ae3f56da57bf895554721951b6).

closed 2017-02-10 by dusty-nv