dusty-nv/ros_deep_learning #2: Failed to load nodelet

vusdo, 2017-08-18:
roslaunch ros_deep_learning imagenet.launch 
WARNING: Catkin package name "jetson-inference" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
... logging to /home/nvidia/.ros/log/29f3cc70-8352-11e7-80d1-00044b8ca462/roslaunch-tegra-ubuntu-1195.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Catkin package name "jetson-inference" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
started roslaunch server http://192.168.144.13:39466/

SUMMARY
========

PARAMETERS
 * /imagenet_node/class_labels_path: /home/nvidia/catk...
 * /imagenet_node/model_path: /home/nvidia/catk...
 * /imagenet_node/prototxt_path: /home/nvidia/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    imagenet_node (nodelet/nodelet)
    standalone_nodelet (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Catkin package name "jetson-inference" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
process[standalone_nodelet-1]: started with pid [1237]
process[imagenet_node-2]: started with pid [1238]
[ INFO] [1503041413.613694631]: Loading nodelet /imagenet_node of type ros_deep_learning/ros_imagenet to manager standalone_nodelet with the following remappings:
[ INFO] [1503041413.613828102]: /imagenet_node/imin -> /camera/rgb/image_rect_color
[ INFO] [1503041413.622413040]: waitForService: Service [/standalone_nodelet/load_nodelet] has not been advertised, waiting...
[ INFO] [1503041413.628775729]: Initializing nodelet with 6 worker threads.
[ INFO] [1503041413.644417973]: waitForService: Service [/standalone_nodelet/load_nodelet] is now available.
/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /home/nvidia/catkin_ws/devel/lib//libros_deep_learning_nodelets.so: undefined symbol: _ZN8imageNet6CreateEPKcS1_S1_S1_S1_S1_j
[FATAL] [1503041413.749550813]: Failed to load nodelet '/imagenet_node` of type `ros_deep_learning/ros_imagenet` to manager `standalone_nodelet'
[standalone_nodelet-1] process has died [pid 1237, exit code 127, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=standalone_nodelet __log:=/home/nvidia/.ros/log/29f3cc70-8352-11e7-80d1-00044b8ca462/standalone_nodelet-1.log].
log file: /home/nvidia/.ros/log/29f3cc70-8352-11e7-80d1-00044b8ca462/standalone_nodelet-1*.log
[imagenet_node-2] process has died [pid 1238, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load ros_deep_learning/ros_imagenet standalone_nodelet ~imin:=/camera/rgb/image_rect_color __name:=imagenet_node __log:=/home/nvidia/.ros/log/29f3cc70-8352-11e7-80d1-00044b8ca462/imagenet_node-2.log].
log file: /home/nvidia/.ros/log/29f3cc70-8352-11e7-80d1-00044b8ca462/imagenet_node-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

vusdo, 2017-08-21:
make install  priviliged mode..

ArtlyStyles, 2018-08-03:
I guess before I compile this node in ROS with catkin_make, I need somehow install and compile the jetson-inference project on TX2. Is that correct?

dusty-nv, 2018-12-21:
@robotlinker if you pull from master again, this should be fixed now in commit [0f5a6b](https://github.com/dusty-nv/ros_deep_learning/commit/0f5a6bcd03eaf1683de8bc65764344ac5ddc7a15)

robotlinker, 2018-12-24:
Thank you! This fixed the problem.