dusty-nv/ros_deep_learning #3: Failed to load nodelet /gst_camera of type [ros_jetson_video/gst_camera]

bpinaya, 2017-09-22:
Hi there @dusty-nv , in the launch file you include a node which I assume is the node that takes care of the gst_camera. Do you plan to release such node?
Otherwise there is this error:
```
[ERROR] [1506071741.153780987]: Failed to load nodelet [/gst_camera] of type [ros_jetson_video/gst_camera] even after refreshing the cache: According to the loaded plugin descriptions the class ros_jetson_video/gst_camera with base class type nodelet::Nodelet does not exist. Declared types are  ros_deep_learning/ros_imagenet
[ERROR] [1506071741.153964994]: The error before refreshing the cache was: According to the loaded plugin descriptions the class ros_jetson_video/gst_camera with base class type nodelet::Nodelet does not exist. Declared types are  ros_deep_learning/ros_imagenet
[FATAL] [1506071741.154335455]: Failed to load nodelet '/gst_camera` of type `ros_jetson_video/gst_camera` to manager `standalone_nodelet'
[gst_camera-4] process has died [pid 32494, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load ros_jetson_video/gst_camera standalone_nodelet ~image_raw:=/image_raw __name:=gst_camera __log:=/home/ubuntu/.ros/log/92f488dc-9f76-11e7-b5c5-00044b633dff/gst_camera-4.log].
log file: /home/ubuntu/.ros/log/92f488dc-9f76-11e7-b5c5-00044b633dff/gst_camera-4*.log

```

dusty-nv, 2017-09-22:
Hmm thanks Benjamin, let me see if I can locate this, otherwise we may need to re-create the nodelet from the gstCamera sources included with jetson-inference.  These nodelets were an experiment to get better performance anyhow.  Turns out we needed to author our own I/O nodelets as well.

You may also be interested, we now have caffe_ros node available under the github.com/NVIDIA-Jetson account.  It loads caffe models with TensorRT and the node portion is more current than what I have here.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

bpinaya, 2017-09-25:
Oh thanks! I'll check that one out. Closing this one.

closed 2017-09-25 by bpinaya

bpinaya, 2017-11-15:
Hey @dusty-nv , did you find the nodelet by any chance? You used the files in the gst-camera from the jetson-inference to build it right? I think I'll try to recreate the nodelet, I tried the jetson caffe_ros repo but it's not working and haven't gotten their launch files with the gscam node to work.

telemething, 2018-08-22:
Did either of you (re)create this nodelet?

bpinaya, 2018-08-23:
Hey there, I ended up doing an implementation myself but can disclose it since it was done for a company, but it basically resumes to get the image from a rostopic, converting it to your format and pass it to the serialized TensorRT Network.
But I think it's better to use the direct gstcam or v4l to get the images.