dusty-nv/ros_deep_learning #5: failed at building ros_deep_learning

Bogonek, 2018-12-26:
Hi everyone.

I'm stuck at building code of ros_deep_learning, just after "catkin_make" I got:

````
nvidia@tegra-ubuntu:~/catkin_ws$` catkin_make
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "cmake /home/nvidia/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install -G Unix Makefiles" in "/home/nvidia/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ros_deep_learning
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ros_deep_learning'
-- ==> add_subdirectory(ros_deep_learning)
-- Found CUDA: /usr/local/cuda-9.0 (found version "9.0") 
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.7") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/nvidia/catkin_ws/build"
####
Scanning dependencies of target imagenet
Scanning dependencies of target ros_deep_learning_nodelets
Scanning dependencies of target detectnet
[ 11%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o
[ 22%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
[ 33%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o
[ 44%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/image_converter.cpp.o
[ 55%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o
[ 66%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/image_converter.cpp.o
[ 77%] Linking CXX executable /home/nvidia/catkin_ws/devel/lib/ros_deep_learning/detectnet
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDeviceNameFromFd2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
ros_deep_learning/CMakeFiles/detectnet.dir/build.make:156: recipe for target '/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/detectnet' failed
make[2]: *** [/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/detectnet] Error 1
CMakeFiles/Makefile2:483: recipe for target 'ros_deep_learning/CMakeFiles/detectnet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/detectnet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 88%] Linking CXX shared library /home/nvidia/catkin_ws/devel/lib/libros_deep_learning_nodelets.so
[ 88%] Built target ros_deep_learning_nodelets
[100%] Linking CXX executable /home/nvidia/catkin_ws/devel/lib/ros_deep_learning/imagenet
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDeviceNameFromFd2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
ros_deep_learning/CMakeFiles/imagenet.dir/build.make:156: recipe for target '/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/imagenet' failed
make[2]: *** [/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/imagenet] Error 1
CMakeFiles/Makefile2:808: recipe for target 'ros_deep_learning/CMakeFiles/imagenet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/imagenet.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

dusty-nv, 2018-12-26:
Your version of openGL has been changed/overwritten/corrupted from the version that came with JetPack-L4T.

See here:   https://devtalk.nvidia.com/default/topic/1044109/jetson-agx-xavier/no-usr-lib-aarch64-linux-gnu-tegra-libgl-so-0-with-jetpack-4-4-1/post/5297052/#5297052
________________________________
From: Bogonek <notifications@github.com>
Sent: Wednesday, December 26, 2018 9:01:03 AM
To: dusty-nv/ros_deep_learning
Cc: Subscribed
Subject: [dusty-nv/ros_deep_learning] failed at building ros_deep_learning (#5)

Hi everyone.

I'm stuck at building code of ros_deep_learning, just after "catkin_make" I got:

nvidia@tegra-ubuntu:~/catkin_ws$` catkin_make
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "cmake /home/nvidia/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install -G Unix Makefiles" in "/home/nvidia/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ros_deep_learning
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ros_deep_learning'
-- ==> add_subdirectory(ros_deep_learning)
-- Found CUDA: /usr/local/cuda-9.0 (found version "9.0")
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.7")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/nvidia/catkin_ws/build"
####
Scanning dependencies of target imagenet
Scanning dependencies of target ros_deep_learning_nodelets
Scanning dependencies of target detectnet
[ 11%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o
[ 22%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
[ 33%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o
[ 44%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/image_converter.cpp.o
[ 55%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o
[ 66%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/image_converter.cpp.o
[ 77%] Linking CXX executable /home/nvidia/catkin_ws/devel/lib/ros_deep_learning/detectnet
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
[... 21 lines cut ...]
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDeviceNameFromFd2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
ros_deep_learning/CMakeFiles/imagenet.dir/build.make:156: recipe for target '/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/imagenet' failed
make[2]: *** [/home/nvidia/catkin_ws/devel/lib/ros_deep_learning/imagenet] Error 1
CMakeFiles/Makefile2:808: recipe for target 'ros_deep_learning/CMakeFiles/imagenet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/imagenet.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

—
You are receiving this because you are subscribed to this thread.
Reply to this email directly, view it on GitHub<https://github.com/dusty-nv/ros_deep_learning/issues/5>, or mute the thread<https://github.com/notifications/unsubscribe-auth/AOpDK4xuhmKXCC-Z0D_ahBpfxjFKvQJAks5u84EfgaJpZM4ZhxyT>.

-----------------------------------------------------------------------------------
This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Bogonek, 2018-12-26:
> Your version of openGL has been changed/overwritten/corrupted from the version that came with JetPack-L4T. See here: https://devtalk.nvidia.com/default/topic/1044109/jetson-agx-xavier/no-usr-lib-aarch64-linux-gnu-tegra-libgl-so-0-with-jetpack-4-4-1/post/5297052/#5297052
> […](#)

But I'm working on TX2 flashed with JetPack 3.3

Tyagi-Ramakrishnan, 2019-02-26:
Hi,

I have a similar issue. Everything was going great with the install until the catkin_make command. I am running on a Jetson TX2. ROS kinetic works fine.

Here is my output:
cd ~/catkin_ws/src
nvidia@tegra-ubuntu:~/catkin_ws/src$ git clone https://github.com/dusty-nv/ros_deep_learning
Cloning into 'ros_deep_learning'...
remote: Enumerating objects: 71, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 149 (delta 43), reused 55 (delta 27), pack-reused 78
Receiving objects: 100% (149/149), 38.47 KiB | 0 bytes/s, done.
Resolving deltas: 100% (65/65), done.
Checking connectivity... done.
nvidia@tegra-ubuntu:~/catkin_ws/src$ cd ../
nvidia@tegra-ubuntu:~/catkin_ws$ catkin_make
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "cmake /home/nvidia/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install -G Unix Makefiles" in "/home/nvidia/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ros_deep_learning
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ros_deep_learning'
-- ==> add_subdirectory(ros_deep_learning)
CMake Error at ros_deep_learning/CMakeLists.txt:14 (find_package):
  By not providing "Findjetson-utils.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "jetson-utils", but CMake did not find one.

  Could not find a package configuration file provided by "jetson-utils" with
  any of the following names:

    jetson-utilsConfig.cmake
    jetson-utils-config.cmake

  Add the installation prefix of "jetson-utils" to CMAKE_PREFIX_PATH or set
  "jetson-utils_DIR" to a directory containing one of the above files.  If
  "jetson-utils" provides a separate development package or SDK, be sure it
  has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Please, help!

dusty-nv, 2019-02-26:
@Tyagi-Ramakrishnan did you run the `sudo make install` step from [this section](https://github.com/dusty-nv/ros_deep_learning#jetson-inference)?

Tyagi-Ramakrishnan, 2019-02-26:
Hi,

Thanks for getting back quickly. Yes, I did. I was able to run the examples from the Jetson Inference git.
It took a while to install I remember that. Do you recommend I run it again?

dusty-nv, 2019-02-26:
Sure, try running the “sudo make install” step again.  This should just copy the required files to /usr, not rebuild everything.

Tyagi-Ramakrishnan, 2019-02-26:
Ok new problem. I restarted the install after getting rid of the old files. I am getting an error on the 'make' step.

nvidia@tegra-ubuntu:~/jetson-inference/build$ make
[ 63%] Built target jetson-utils
[ 63%] Built target jetson-inference
[ 63%] Linking CXX executable ../aarch64/bin/imagenet-console
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDeviceNameFromFd2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
imagenet-console/CMakeFiles/imagenet-console.dir/build.make:102: recipe for target 'aarch64/bin/imagenet-console' failed
make[2]: *** [aarch64/bin/imagenet-console] Error 1
CMakeFiles/Makefile2:123: recipe for target 'imagenet-console/CMakeFiles/imagenet-console.dir/all' failed
make[1]: *** [imagenet-console/CMakeFiles/imagenet-console.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Tyagi-Ramakrishnan, 2019-02-26:
fixed the last comment's issue with this link: https://devtalk.nvidia.com/default/topic/946136/building-an-opengl-application/

Tyagi-Ramakrishnan, 2019-02-26:
I was able to compile catkin_make successfully! Thanks for the help!

closed 2019-10-21 by dusty-nv