dusty-nv/ros_deep_learning #11: Segnet launchfile

ArtlyStyles, 2019-07-02:
Hi, Could anyone provide a launch file for segnet?

FPSychotic, 2019-11-08:
same here, if it is  possible with a example for beginners as me
thank you

r91andersson, 2020-03-09:
This would be really helpful! Any plan for fixing this?

mcdzim, 2020-04-29:
Hi, if possible could I add support to asking for another launch file but also has anyone managed to make a launchfile for detectnet?

mcdzim, 2020-04-30:
Got Detectnet working with 

<launch>
    <rosparam  file="$(find ziva_bringup)/config/detectnet.yaml" command="load"/>

    <node name="detectnet"          pkg="ros_deep_learning"  type="detectnet" output="screen">>
        <remap from="~image_in" to="/webcam/image_raw"/>
    </node>

</launch>