dusty-nv/ros_deep_learning #13: vision_msgs/Detection2DArray.h can not be found

deeprun, 2019-07-20:
When I build it on TX2, I got the following errors:

/home/nvidia/racecar-ws/src/ros_deep_learning/src/node_segnet.cpp:27:36: fatal error: vision_msgs/VisionInfo.h: No such file or directory
compilation terminated.
/home/nvidia/racecar-ws/src/ros_deep_learning/src/node_detectnet.cpp:26:42: fatal error: vision_msgs/Detection2DArray.h: No such file or directory

How can I fix it?

dusty-nv, 2019-07-21:
Hi Deeprun, which version of ROS do you have installed?

Thanks,
Dusty

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ArtlyStyles, 2019-07-21:
I am using Kinetic

deeprun, 2019-07-22:
Hi Dusty:

I am using Kinetic

laljarus, 2019-07-23:
Hi 
I get a similar error while building the package.  The error says ros cannot find package vision_msgs.cmake. Looks to me like some ros package is missing but I cannot find it. 
Could you please help to solve it
Best Regards
Suraj