dusty-nv/ros_deep_learning #14: catkin_make failed after jetson-inference update

aparwal, 2019-07-23:
Could not build ros package after installing latest version of jetson-inference

Output of catkin_make:

<pre><b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:</b> In function ‘<b>void img_callback(const ImageConstPtr&amp;)</b>’:
<b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:74:119:</b> <font color="#EF2929"><b>error: </b></font>no matching function for call to ‘<b>detectNet::Detect(float*, uint32_t, uint32_t, float*&amp;, int*, float*&amp;)</b>’
  const bool result = net-&gt;Detect(cvt-&gt;ImageGPU(), cvt-&gt;GetWidth(), cvt-&gt;GetHeight(), bbCPU, &amp;numBoundingBoxes, confCPU);
<font color="#8AE234"><b>                                                                                                                       ^</b></font>
In file included from <b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:29:0</b>:
<b>/usr/local/include/jetson-inference/detectNet.h:271:6:</b> <font color="#34E2E2"><b>note: </b></font>candidate: int detectNet::Detect(float*, uint32_t, uint32_t, detectNet::Detection**, uint32_t)
  int Detect( float* input, uint32_t width, uint32_t height, Detection** detections, uint32_t overlay=OVERLAY_BOX );
<font color="#8AE234"><b>      ^</b></font>
<b>/usr/local/include/jetson-inference/detectNet.h:271:6:</b> <font color="#34E2E2"><b>note: </b></font>  candidate expects 5 arguments, 6 provided
<b>/usr/local/include/jetson-inference/detectNet.h:283:6:</b> <font color="#34E2E2"><b>note: </b></font>candidate: int detectNet::Detect(float*, uint32_t, uint32_t, detectNet::Detection*, uint32_t)
  int Detect( float* input, uint32_t width, uint32_t height, Detection* detections, uint32_t overlay=OVERLAY_BOX );
<font color="#8AE234"><b>      ^</b></font>
<b>/usr/local/include/jetson-inference/detectNet.h:283:6:</b> <font color="#34E2E2"><b>note: </b></font>  candidate expects 5 arguments, 6 provided
<b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:77:7:</b> <font color="#EF2929"><b>error: </b></font>in argument to unary !
  if( !result )
<font color="#8AE234"><b>       ^</b></font>
<b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:</b> In function ‘<b>int main(int, char**)</b>’:
<b>/home/nvidia/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:205:20:</b> <font color="#EF2929"><b>error: </b></font>‘<b>class detectNet</b>’ has no member named ‘<b>GetMaxBoundingBoxes</b>’
  maxBoxes   = net-&gt;GetMaxBoundingBoxes();  
<font color="#8AE234"><b>                    ^</b></font>
ros_deep_learning/CMakeFiles/detectnet.dir/build.make:62: recipe for target &apos;ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o&apos; failed
make[2]: *** [ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o] Error 1
CMakeFiles/Makefile2:7449: recipe for target &apos;ros_deep_learning/CMakeFiles/detectnet.dir/all&apos; failed
make[1]: *** [ros_deep_learning/CMakeFiles/detectnet.dir/all] Error 2
Makefile:138: recipe for target &apos;all&apos; failed
make: *** [all] Error 2
<font color="#CC0000">Invoking </font><font color="#EF2929"><b>&quot;make -j6 -l6&quot;</b></font><font color="#CC0000"> failed</font>
</pre>

Seems like signature of detectNet::Detect() changed after [this commit of jetson-inference](https://github.com/dusty-nv/jetson-inference/commit/62add6fbf3bb888537979cd430f29252b9c3e498l) but this package is using the old signature. 
Is there a tag/commit of jetson-inference for which this package works?

dusty-nv, 2019-07-23:
Hi @aparwal, sorry - looks like I need to update this ROS node.

For now you should be able to use the [`L4T-R32.1`](https://github.com/dusty-nv/jetson-inference/tree/L4T-R32.1) tag of jetson-inference, that was captured before these changes were merged into master.

aparwal, 2019-07-24:
Thanks, the node works for that tag.

agoila, 2019-07-27:
I am sorry, but I am relatively new to GitHub, and have been having difficulty understanding what you mean by `tag`? Is this the same as `branch`?

I ran into the same issue last night, and am wondering if I need to checkout the branch `L4T-R32.1`, and follow the build steps again (for jetson-inference) to get this node to work?

Thanks for your help @dusty-nv @aparwal !

luo980, 2019-07-31:
Found the same problem, but I will wait for the update :D
Thanks

SimoGira, 2019-08-01:
Looking at Detect method in "detectNet.h" and the same Detect method used in "node_detectnet.cpp", jetson-inference repo doesn't seems "aligned " with the ros_deep_learninig repo.
I'm waiting for the update too!! ;)

neurondizzy, 2019-09-03:
Hello kmartin .... just used your update and all went well with build, waiting on a camera to move further. Thanks for the update.

dusty-nv, 2019-10-21:
The `ros_deep_learning` repo has now been updated for JetPack 4.2 and to build against `jetson-inference` master.  Let me know if you have any more problems building it.

closed 2019-10-21 by dusty-nv