dusty-nv/ros_deep_learning #15: Jetson-nano

knoob-noob, 2019-07-31:
Would this work with jetson nano? Jetpack and jetson inference provide support for nano. If after installing them, I try to build ros_deep_learning package on jetson nano, would it work?

dusty-nv, 2019-07-31:
Yes, install the L4T-R32.1 branch of jetson-inference and it should work

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

knoob-noob, 2019-08-01:
Hey! Thanks for the reply.
These are the steps I followed-
First I installed jetpack on jetson nano.
Then, I cloned the master repo and followed the appropriate steps. After this, I was able to use jetson inference to do image recognition and object detection.
Then, I tried building the ros_deep_learning package but got an error while doing catkin_make.

Then, I found out that I'm supposed to be installing the L4t-R32.1 branch, so I deleted the jetson-inference folder and installed the L4T branch, and after this, I couldn't do image recognition or detection. Getting the following errors-
**imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet**

Can you please tell how to resolve these issues? And is there a possibility that I'd have to install jetpack again from scratch?

Thanks.

knoob-noob, 2019-08-02:
Also, can you please tell me whether I should clone L4TR32.1 branch using this command -
**git clone -b L4T-R32.1 https://github.com/dusty-nv/jetson-inference**
or is this not the proper way to clone here?

Because after following the steps mentioned at **https://github.com/dusty-nv/ros_deep_learning**, (but using the above command to clone the repo), when I run cmake ../ and sudo make install,it's not downloading the models on it's own, and when the make is complete, the model downloader file (download-models.sh) along with some other files is missing from the tools directory.

How do I fix these issues?

dusty-nv, 2019-10-21:
The `ros_deep_learning` repo has now been updated for JetPack 4.2 and to build against `jetson-inference` master.  Let me know if you have any more problems building it.

closed 2019-10-21 by dusty-nv