dusty-nv/ros_deep_learning #16: failed to convert 1280x720 bgra8 image

JorgeVilchis, 2019-08-29:
I using ZEDmini on a Jetson TX2, while trying to use the the image topic /zed/rgb/image_rect_color I am getting the error:

failed to convert 1280x720 bgra8 image 

I have tried to subscribe the ros_deep_learning node to other zed camera image topics and same error occurs.
------- 

[cuda]  cudaAllocMapped 131072 bytes, CPU 0x102540000 GPU 0x102540000
[cuda]  cudaAllocMapped 32768 bytes, CPU 0x102350000 GPU 0x102350000
[ INFO] [1567098659.557982573]: model hash => 965427319687731864
[ INFO] [1567098659.558040077]: hash string => /usr/local/bin/networks/ped-100/snapshot_iter_70800.caffemodel/usr/local/bin/networks/ped-100/class_labels.txt
[ INFO] [1567098659.560447145]: node namespace => /detectnet
[ INFO] [1567098659.560522857]: class labels => /detectnet/class_labels_965427319687731864
[ INFO] [1567098659.571631642]: detectnet node initialized, waiting for messages
[ INFO] [1567098659.736026296]: converting 1280x720 bgra8 image
[ERROR] [1567098659.736376120]: 1280x720 image is in bgra8 format, expected bgr8
[ INFO] [1567098659.736437304]: failed to convert 1280x720 bgra8 image
[ INFO] [1567098659.768120524]: converting 1280x720 bgra8 image

JorgeVilchis, 2019-08-30:
Update: I created a dropbox folder containing a ROS bag . If ros_deep_learning node is subscribed  to the topic /rgb_image the issue mentioned above can  be reproduced.

Setup info: 
Jetson TX2
ROS kinetic
Ubuntu 16.04

Link where ROS bag can be downloaded: 

https://www.dropbox.com/s/bkcxj68dr38j50c/video_bag.tar.gz 

Any help will be highly appreciate it.

dipampatel18, 2019-09-24:
> I am using ZEDmini on a Jetson TX2, while trying to use the the image topic /zed/rgb/image_rect_color I am getting the error:
> 
> failed to convert 1280x720 bgra8 image
> 
> ## I have tried to subscribe the ros_deep_learning node to other zed camera image topics and same error occurs.
> [cuda] cudaAllocMapped 131072 bytes, CPU 0x102540000 GPU 0x102540000
> [cuda] cudaAllocMapped 32768 bytes, CPU 0x102350000 GPU 0x102350000
> [ INFO] [1567098659.557982573]: model hash => 965427319687731864
> [ INFO] [1567098659.558040077]: hash string => /usr/local/bin/networks/ped-100/snapshot_iter_70800.caffemodel/usr/local/bin/networks/ped-100/class_labels.txt
> [ INFO] [1567098659.560447145]: node namespace => /detectnet
> [ INFO] [1567098659.560522857]: class labels => /detectnet/class_labels_965427319687731864
> [ INFO] [1567098659.571631642]: detectnet node initialized, waiting for messages
> [ INFO] [1567098659.736026296]: converting 1280x720 bgra8 image
> [ERROR] [1567098659.736376120]: 1280x720 image is in bgra8 format, expected bgr8
> [ INFO] [1567098659.736437304]: failed to convert 1280x720 bgra8 image
> [ INFO] [1567098659.768120524]: converting 1280x720 bgra8 image

@JorgeVilchis @dusty-nv I am facing the same issue. Could you find any solution to this? I am stuck in this same problem since a long time. I would really appreciate if you could provide me any help. Thanks!

vxgu86, 2019-10-08:
@JorgeVilchis @dipampatel18 @dusty-nv did you solve this problem?
cause I am meeting the same one..

qboticslabs, 2019-10-11:
@vxgu86 @dipampatel18 @dusty-nv There is a tempory fix for this issue. Write a new ROS node which is subscribing to the image you want to feed into the deep learning node, and in the image callback, you can do the following conversion mentioned in the code snippet, and publish the converted image to another topic. 

Feed that converted image as the input of deep learning node. This fix worked for me, I can't share the complete code, but I can share the conversion snippet in the callback.

void imageCallback(const sensor_msgs::ImageConstPtr& msg)
{
  try
  {
  
    frame = cv_bridge::toCvShare(msg, "bgr8")->image;

   sensor_msgs::ImagePtr msg;      
   msg = cv_bridge::CvImage(std_msgs::Header(), "bgr8", frame).toImageMsg();
   pub_converted.publish(msg);    
   
 

  }
  catch (cv_bridge::Exception& e)
  {
    ROS_ERROR("Could not convert from '%s' to 'bgr8'.", msg->encoding.c_str());
  }
}