dusty-nv/ros_deep_learning #18: Failed at catkin_make detectnet

bj-neilson, 2019-09-04:
Hi,
I've gotten most of the way through building ros deep learning.
Using latest Nano image (r32.2), Ubuntu 18.04, CUDA 10.0, ROS Melodic.
Getting the following...

nano@nano:~/catkin_ws$ catkin_make
Base path: /home/nano/catkin_ws
Source space: /home/nano/catkin_ws/src
Build space: /home/nano/catkin_ws/build
Devel space: /home/nano/catkin_ws/devel
Install space: /home/nano/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nano/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/nano/catkin_ws/build"
####
[ 50%] Built target segnet
[ 50%] Built target imagenet
[ 58%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o
[ 83%] Built target ros_deep_learning_nodelets
/home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp: In function ‘void img_callback(const ImageConstPtr&)’:
/home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:74:119: error: no matching function for call to ‘detectNet::Detect(float*, uint32_t, uint32_t, float*&, int*, float*&)’
  const bool result = net->Detect(cvt->ImageGPU(), cvt->GetWidth(), cvt->GetHeight(), bbCPU, &numBoundingBoxes, confCPU);
                                                                                                                       ^
In file included from /home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:29:0:
/usr/local/include/jetson-inference/detectNet.h:271:6: note: candidate: int detectNet::Detect(float*, uint32_t, uint32_t, detectNet::Detection**, uint32_t)
  int Detect( float* input, uint32_t width, uint32_t height, Detection** detections, uint32_t overlay=OVERLAY_BOX );
      ^~~~~~
/usr/local/include/jetson-inference/detectNet.h:271:6: note:   candidate expects 5 arguments, 6 provided
/usr/local/include/jetson-inference/detectNet.h:283:6: note: candidate: int detectNet::Detect(float*, uint32_t, uint32_t, detectNet::Detection*, uint32_t)
  int Detect( float* input, uint32_t width, uint32_t height, Detection* detections, uint32_t overlay=OVERLAY_BOX );
      ^~~~~~
/usr/local/include/jetson-inference/detectNet.h:283:6: note:   candidate expects 5 arguments, 6 provided
/home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:77:7: error: in argument to unary !
  if( !result )
       ^~~~~~
/home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp: In function ‘int main(int, char**)’:
/home/nano/catkin_ws/src/ros_deep_learning/src/node_detectnet.cpp:205:20: error: ‘class detectNet’ has no member named ‘GetMaxBoundingBoxes’
  maxBoxes   = net->GetMaxBoundingBoxes();
                    ^~~~~~~~~~~~~~~~~~~
ros_deep_learning/CMakeFiles/detectnet.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o] Error 1
CMakeFiles/Makefile2:1436: recipe for target 'ros_deep_learning/CMakeFiles/detectnet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/detectnet.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Any thoughts Dusty?  I've searched for answers and tried installing several times.  Keep getting stuck at this point.

Thanks,
Ben

aparwal, 2019-09-09:
see #14 and #17

dusty-nv, 2019-10-21:
The `ros_deep_learning` repo has now been updated for JetPack 4.2 and to build against `jetson-inference` master.  Let me know if you have any more problems building it.

closed 2019-10-21 by dusty-nv