dusty-nv/ros_deep_learning #19: Using realsense node output for detecting object using detectnet

inderpreetsingh01, 2019-09-04:
Hi @dusty-nv
I am using Realsense D415 camera along with jetson Tx2.
I have launched the camera node using 

`roslaunch realsense2_camera rs_camera.launch `

after that I ran detectnet node using 

`rosrun ros_deep_learning detectnet /detectnet/image_in:=camera/color/image_raw _model_name:=pednet`

after this I am getting the error

failed to convert 640*480 rgb8 image
converting 640*480 rgb8 image
640*480 image is in rgb8 format, expected bgr8

inderpreetsingh01, 2019-09-05:
I solved the issue by writing a separate node for converting raw_image from rgb8 to gbr8 format, by subscribing to the /camera/color/image_raw topic of realsense and publishing it to new topic to which detectnet node will subscribe.

closed 2019-09-05 by inderpreetsingh01

vxgu86, 2019-10-08:
how did you solve this, could you share your solution?
 thanks
@inderpreetsingh01

inderpreetsingh01, 2019-10-09:
Hi @vxgu86 below I am attaching a file which I am using as a ros node.
I hope it helps
[rgb2bgr.txt](https://github.com/dusty-nv/ros_deep_learning/files/3707140/rgb2bgr.txt)

debackerj93, 2020-05-21:
@inderpreetsingh01 Thanks for sharing your solution. I'm having some trouble getting the include files to work.. can you share your insight to this catkin_make problem I get when I try to build the source file:

```
In file included from /home/lduser/catkin_ws/src/ros_deep_learning/src/rgb2bgr.cpp:3:0:
/opt/ros/melodic/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp: No such file or directory
 #include <opencv2/core/core.hpp>
```

I've installed the ros package vision_opencv.. but it isn't finding any of the header files. Sorry, I'm very new to this.

Thanks!