dusty-nv/ros_deep_learning #24: image is in rgb8 format, expected bgr8,

vxgu86, 2019-10-08:
I am getting image_raw from a gscam drived camera, with /camera/image_raw subscribed.
the detectNet error with the following:

[cuda]  cudaAllocMapped 131072 bytes, CPU 0x101c30000 GPU 0x101c30000
[cuda]  cudaAllocMapped 32768 bytes, CPU 0x101b40000 GPU 0x101b40000
[ INFO] [1570526393.448824707]: model hash => 965427319687731864
[ INFO] [1570526393.448906165]: hash string => /usr/local/bin/networks/ped-100/snapshot_iter_70800.caffemodel/usr/local/bin/networks/ped-100/class_labels.txt
[ INFO] [1570526393.452781894]: node namespace => /detectnet
[ INFO] [1570526393.452866842]: class labels => /detectnet/class_labels_965427319687731864
[ INFO] [1570526393.467713092]: detectnet node initialized, waiting for messages
[ INFO] [1570526393.722328769]: converting 1920x1080 rgb8 image
[ERROR] [1570526393.722481165]: 1920x1080 image is in rgb8 format, expected bgr8
[ INFO] [1570526393.722541269]: failed to convert 1920x1080 rgb8 image
[ INFO] [1570526393.922863144]: converting 1920x1080 rgb8 image
[ERROR] [1570526393.923110279]: 1920x1080 image is in rgb8 format, expected bgr8
[ INFO] [1570526393.923377988]: failed to convert 1920x1080 rgb8 image

how can I change the image_encodings???

cocodirasta, 2019-10-30:
Hi, 

Have you found a way to solve your issue ? 
I have the same problem.

Thank you

gclarke42, 2020-03-11:
It looks like an in code fix for this is in progress but incase you need a faster solution. I was able to make a sub/pub node using cv_bridge which converted the rgb to bgr and then republish the new image.  Not an ideal solution but was pretty simple.  The cv_bridge example was basically what I used for the node. http://wiki.ros.org/cv_bridge

aveen-d, 2020-04-06:
> It looks like an in code fix for this is in progress but incase you need a faster solution. I was able to make a sub/pub node using cv_bridge which converted the rgb to bgr and then republish the [...]

hi i am facing the same error. can you please tell me where in the code can i fix this issue? I am new to this and i am not able to figure it out