dusty-nv/ros_deep_learning #25: This repo does not compile

deeprun, 2019-10-10:
Could you update? This repo is no longer sync with jetson-inference. This repo does not compile

dusty-nv, 2019-10-10:
Sorry about that, I will fix it.  For now, please use the L4T-R32.1 branch of jetson-inference repo.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2019-10-21:
The repo has now been updated for the latest JetPack and compiles against `jetson-inference` master.  Let me know if you have any more issues building.

closed 2019-10-21 by dusty-nv