dusty-nv/ros_deep_learning #28: anyway to visualize the detect result like in jet-inference?

vxgu86, 2019-11-03:
anyway to visualize the detect result like in jet-inference?

gclarke42, 2020-03-11:
I'm looking to do the same, have you or anyone else figured out how to do this?  Would like to be able to republish the image with the bounding box on it along with the detection topic.

matthaeusheer, 2020-03-19:
@gclarke42 
Thanks for the snippets. I think what is missing is the declaration of image_pub in node_detectnet.cpp:  
```image_transport::Publisher* image_pub = NULL;```  

When visualizing the /camera/image_det_output in rviz using detectNet::OVERLAY_BOX as an overlay, the conversion crashes. At the beginning I can see some blue box flickering shortly sometimes where the detections should be, then I get 
```
[ERROR] [1584626931.491593779]: back-conversion: failed to convert 640x480 image with CUDA
[ERROR] [1584626931.500285776]: conversion: failed to convert 640x480 image with CUDA
[ERROR] [1584626931.533418376]: conversion: failed to convert 640x480 image with CUDA
...
```

Any idea how to fix this? Appreciate your work. It does not crash when using OVERLAY_NONE but then the whole point is kind of gone. As a work around one could probably draw directly on the back-converted image using openCV.

gclarke42, 2020-03-19:
I updated the first response with a couple of missing lines, the first is the ptr you mentioned, the 2nd is a line change that enables the bounding box in img_detect..

It sounds like you had both of those missing lines covered already, the only difference i see jump out is I used all 3 bounding box options instead of just the plain box.  I'm not sure about the errors, just based on the message it seems to be having a problem running the rgb-bga conversion in CUDA.  I'm not well versed in CUDA, when writing that function i basically just reversed everything that it did in the initial conversion in order to copy the modified image from cvt back into an image ptr.  RVIZ should be able to process the BGR image i think, so you might be able to skip that step and instead just copy the updated image to the stream ptr.

matthaeusheer, 2020-03-20:
> I did notice that with some models, moving the camera around a bunch or having too many detentions caused a crash
Same here, which is pretty much a no-go for my application (running on a drone) :)  

I did a workaround by simply drawing detection boxes using openCV, [my ros_deep_learning fork](https://github.com/matthaeusheer/ros_deep_learning/blob/feat/detect_img_pub/src/node_detectnet.cpp), which works just fine.
So far I was running on my laptop (Dell XPS15, GTX960M, Ubuntu 18.04 with 720p usb-webcam) but I will test it on the Jetson TX2 as well to figure out if there is some difference. I did use facenet and pednet networks. In my case I published the images from video_stream_opencv but I think ones ros_deep_learning has the images this should not be of relevance.

Let's see, @dusty-nv, do you have some hints on this one?

jamesthesken, 2020-05-06:
Thank you @gclarke42 those code snippets worked on the Nano. Will report back on some testing or if I come across any other ideas