dusty-nv/ros_deep_learning #33: loading my own onxx model for use with ros_deep_learning

BHamlen, 2020-04-12:
hi @dusty-nv 

Hope all is well! I trained a model on darknet, yolov3 with my own dataset, then converted that to a .onnx file. How can I load that model to use with ros_deep_learning?

fwarmuth, 2020-05-04:
For me it worked with a little fiddling...
it seems that the parameter 'parsing'-block handles a `_model_path:=<path to onnx>` not right, at least in my case i needed to overwrite 'use_model_name` to  `false' so that the node used my onnx model provided by `rosrun ros_deep_learning detectnet /detectnet/image_in:=/my/cam/image/topic _model_path:=/path/to/file.onnx`. 

Unfortunately my converted Yolov3-tiny results in:
```
[ INFO] [1588593271.498076451]: converting 1280x640 bgr8 image
[TRT]   detectNet::Detect() -- ONNX -- coord (-inf, -inf) (-inf, -inf)  image 1280x640
[ INFO] [1588593271.501776997]: detected 1 objects in 1280x640 image
object 0 class #0 (class #0)  confidence=1.000000
object 0 bounding box (-inf, -inf)  (-inf, -inf)  w=nan  h=nan
```
seems that something else is wrong...
@BHamlen  did you menage to use the detectnet node with a custom network? i would love to know how!
@dusty-nv any tipps on the corrupt detection output? what did i potentially mess up this time?

regards

BHamlen, 2020-05-04:
@fwarmuth unfortunately not :(
My original model was trained on darknet, and I found a helpful package called [darknet_ros](https://github.com/leggedrobotics/darknet_ros), which allowed me to use the .cfg and .weights from darknet with ROS on my xavier. Not sure if you're wanting to use a darknet model, but if so hopefully that helps!

fwarmuth, 2020-05-05:
@BHamlen How is the performance? i want to use tensort for higher FPS.

BHamlen, 2020-05-05:
@fwarmuth Performance on the xavier with a 720p webcam is about 15-20fps. I'm sure tensorRT will result in higher FPS, yeah. What is your application?

fwarmuth, 2020-05-07:
@BHamlen Hey, my application is an Safety AGV, some autonomous moving thing with sensors :D. But unfortently i can NOT get the tensorRT things to work with my retrained networks....
Same as you i will try now the darknet_ros node. It works on my dev machine but i cant get to to compile on the Xavier, seems to be an opencv issue. What version of opencv are you using?

BHamlen, 2020-05-07:
ah yeah I was having problems with opencv versions too. I went through a couple attempts with different versions so I dont remember exactly, but I believe having opencv 4.x was what ended up working