dusty-nv/ros_deep_learning #34: multiple cameras

aveen-d, 2020-04-17:
Hi , I wanted to know how do I launch detectnet node for multiple camera images at the same time. I have 4 camera image topics with me but how can I run object detection node on all the images at the same time. I am a beginner to this , can you please help me?

Thanks in advance

r91andersson, 2021-05-03:
What I've found out to work was to modify the code inside node_detectnet.cpp.
In my case I have 3 camera live feed I want to run detection on, so what I did was to add 2 more subscribers (basically just copy paste the first subscriber).
`	ROS_GET_PARAMETER("topic_image_in_camera_1", topic_image_in_camera_1);
	ROS_GET_PARAMETER("topic_image_in_camera_2", topic_image_in_camera_2);
	ROS_GET_PARAMETER("topic_image_in_camera_3", topic_image_in_camera_3);`

Then,

`	/*
	 * subscribe to image topic
	 */
	auto img_sub_1 = ROS_CREATE_IMAGE_TRANSPORT_SUBSCRIBER(sensor_msgs::Image, topic_image_in_camera_1, 5, img_callback_camera_1);
	auto img_sub_2 = ROS_CREATE_IMAGE_TRANSPORT_SUBSCRIBER(sensor_msgs::Image, topic_image_in_camera_2, 5, img_callback_camera_2);
	auto img_sub_3 = ROS_CREATE_IMAGE_TRANSPORT_SUBSCRIBER(sensor_msgs::Image, topic_image_in_camera_3, 5, img_callback_camera_3);`

They will all use the same model to run the inference on. i.e they all call this:
`const int numDetections = net->Detect(input_cvt->ImageGPU(), input_cvt->GetWidth(), input_cvt->GetHeight(), &detections, detectNet::OVERLAY_NONE);`

However, I'm not sure if there is any guarantee that a specific callback will have the resource to the detection model and be able to run the inference? Maybe you know this @dusty-nv. Will it be a blocking/wait call if the resource (detection model) is already taken by another thread (callback)?

dusty-nv, 2021-05-03:
> Will it be a blocking/wait call if the resource (detection model) is already taken by another thread (callback)?
Yes detectNet::Detect() is a blocking call

r91andersson, 2021-05-03:
Ok, so if one thread is using the model and blocking the resource, will the other thread wait until the resource becomes free, or will it continue execute?

dusty-nv, 2021-05-03:
Another thread should be able to use it, but I haven't tested that scenario.