dusty-nv/ros_deep_learning #40: Anyway to view detectnet outputs??

aditjha, 2020-07-01:
Hello,
This repo is amazing, thank you so much. Is there any way to view the resulting output detections from the detectnet ros node with the bounding boxes and label, I would really appreciate any help.

k3street, 2020-09-02:
You can subscribe to the node generated see: ros2 topic echo /detectnet/detections

dusty-nv, 2020-09-02:
I also recently added the `/detectnet/overlay` topic.  When subscribed to, it will output the overlay image that has the rendered bounding boxes and labels.

https://github.com/dusty-nv/ros_deep_learning#detectnet-node-1

k3street, 2020-09-02:
I think your detections array is now null in the node topic detections. publisher

dusty-nv, 2020-09-02:
It is only sent when there are actually detections detected in the image

k3street, 2020-09-02:
Hey Dusty, take a look at this. https://youtu.be/VVjh_FYc2Us Also when i took your update the package is now freezing when it wasn't before. I haven't done any triage yet but I will let you know what I find. Also the youtube link I provided doesn't have any sound.
https://youtu.be/VVjh_FYc2Us

dusty-nv, 2020-09-02:
Hmm ok - what version of ROS are you using?  If it's ROS Melodic, you might want to revert to the `ros1` branch of this repo.

Here is where the detection array is populated:  https://github.com/dusty-nv/ros_deep_learning/blob/ac40e93413f4b7cb911a18c0e4d5daac479234d4/src/node_detectnet.cpp#L144

You might want to try adding some additional logging to debug.  But it seems like since you are already getting those detection logging messages each frame, that this array is being filled on the publisher side.

Does it freeze/hiccup even when your screen capture tool isn't running?

k3street, 2020-09-02:
It freezes without the screen capture tool running. I am using ROS2 Eloquent

k3street, 2020-09-02:
I should clarify that this behavior began when I took the latest ros2 update today. There might be something going on there but I don’t have the time to figure it out today. If I come up with a cause I will update the forum.