dusty-nv/ros_deep_learning #45: VideoViewer ROS2 Command returning error (running ROS2 Dashing)

kylelscott, 2020-09-10:
When I run the following command for the VideoViewer demo, I get a "malformed launch argument" error:

`~/ros2_workspace/src$ ros2 launch ros_deep_learning imagenet.ros2.launch input:=csi://0 output:=display://0
malformed launch argument 'imagenet.ros2.launch', expected format '<name>:=<value>'
`

 'I've validated that my cameras are indeed working, and have even tried the similar commands with the imagenet and detectnet node, they both return the same error. Looking for help on what that may be. I am new to ROS and ROS2, but I do understand the basics and after checking the contents of the launch file I believe these are the right parameter values for the command. However, because I am getting an error, I know I am missing something.Thank you.

dusty-nv, 2020-09-10:
Hmm...had you also installed these packages:

``` bash
$ sudo apt-get install ros-eloquent-launch-xml ros-eloquent-launch-yaml
```

kylelscott, 2020-09-10:
I did not. I am running dashing however, and I saw that setting parameters directly from the command line isn't supported. I suppose that's my problem. I wanted to use a distribution that will be supported longer, I can try installing foxy and retry. However, are those launch files you sent already installed when you run the following:
`sudo apt install ros-eloquent-desktop`

kylelscott, 2020-09-10:
I just uninstalled my ROS2 Dashing distribution and installed Eloquent. Followed all the steps to clone ros_deep_learning and ran the command you provided for the xml and yaml installations and the video_viewer launch command still gives me the same error

dusty-nv, 2020-09-10:
Hmm I wonder if its related to the package.  What if you skip the ros_deep_learning and instead specify the path to the .launch file?

kylelscott, 2020-09-10:
`ros2 launch ~/ros2_workspace/src/ros_deep_learning/launch/video_viewer.ros2.launch input:=csi://0 output:=display://0`

Like this?

dusty-nv, 2020-09-10:
I think so, yes

kylelscott, 2020-09-10:
running that returned that error (in the second code snippet). Its saying "input:=cis://0" is an unrecognized file, and says it was not found in the directory. "input:cis://" is the parameter though.

dusty-nv, 2020-09-11:
I think you mean csi://0

Is it launching now, but just not starting the camera?

If ros2 launch is still failing to parse the .launch at all, try removing the := arguments from your command line.   Those can be set in the file for the time being.

kylelscott, 2020-09-11:
Hello, thanks for the response. Sorry I had to step away for a second. But running the command without the arguments returns the same error.

with the arguments:
`kylel@kylel-desktop:~/ros2_workspace/src$ ros2 launch ros_deep_learning video_viewer.ros2.launch input:=csi://0 output:=display://0
malformed launch argument 'video_viewer.ros2.launch', expected format '<name>:=<value>'`

without the arguments:
`kylel@kylel-desktop:~/ros2_workspace/src$ ros2 launch ros_deep_learning video_viewer.ros2.launch
malformed launch argument 'video_viewer.ros2.launch', expected format '<name>:=<value>`

Kinchul, 2020-11-16:
Hey, I had the same problem and I solved it by updating the $AMENT_PREFIX_PATH variable. Somehow, it was not updated when I built my docker image.

I added:
"export AMENT_PREFIX_PATH=/workspace/install/ros_deep_learning:$AMENT_PREFIX_PATH" to my entrypoint.sh

Before:      AMENT_PREFIX_PATH=/opt/ros/eloquent
After:        AMENT_PREFIX_PATH=/workspace/install/ros_deep_learning:/opt/ros/eloquent

gcordova19, 2021-06-30:
I add AMENT_PREFIX_PATH= /home/ros2_example_ws/install/ros_deep_learning:/opt/ros/foxy on ~/.bashrc
Then on ~/ros2_example_ws run 
ros2 launch ros_deep_learning video_viwer.ros2.launch input:="/home/gcordova/9f1de61c-3123-4e27-bc89-3a317bb6973a.mp4" output:=display://0 input_code:=h264

and I got
file 'video_viwer.ros2.launch' was not found in the share directory of package 'ros_deep_learning' which is at '/home/gcordova/ros2_example_ws/install/ros_deep_learning/share/ros_deep_learning'

Someone could help me?