dusty-nv/ros_deep_learning #46: jetson-utils/imageFormat.h: No such file or directory

fnasarias, 2020-09-17:
Encountered this error when building using `catkin_make`. Using ROS Melodic.

`#### Running command: "make cmake_check_build_system" in "/home/pred/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/pred/catkin_ws/build"
####
[  8%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
[ 17%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/node_video_output.cpp.o
[ 17%] Building CXX object ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o
[ 17%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:24:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/segnet.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o] Error 1
CMakeFiles/Makefile2:535: recipe for target 'ros_deep_learning/CMakeFiles/segnet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/segnet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp:8:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 21%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/node_imagenet.cpp:24:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/imagenet.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o] Error 1
CMakeFiles/Makefile2:1542: recipe for target 'ros_deep_learning/CMakeFiles/imagenet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/imagenet.dir/all] Error 2
[ 26%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/image_converter.cpp.o
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/node_video_output.cpp:24:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/video_output.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/video_output.dir/src/node_video_output.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/video_output.dir/src/node_video_output.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 30%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/ros_compat.cpp.o
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.cpp:23:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/build.make:86: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o] Error 1
CMakeFiles/Makefile2:1148: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all] Error 2
In file included from /home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.cpp:23:0:
/home/pred/catkin_ws/src/ros_deep_learning/src/image_converter.h:27:10: fatal error: jetson-utils/imageFormat.h: No such file or directory
 #include <jetson-utils/imageFormat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_deep_learning/CMakeFiles/video_output.dir/build.make:86: recipe for target 'ros_deep_learning/CMakeFiles/video_output.dir/src/image_converter.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/video_output.dir/src/image_converter.cpp.o] Error 1
CMakeFiles/Makefile2:1409: recipe for target 'ros_deep_learning/CMakeFiles/video_output.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/video_output.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

dusty-nv, 2020-09-17:
When you built jetson-inference, did you run `sudo make install` ?

If that doesn't fix it, you may want to re-clone / re-build jetson-inference repo to get the latest.

fnasarias, 2020-09-25:
Re-cloning and re-building jetson-inference worked! Thanks so much!