dusty-nv/ros_deep_learning #53: Could be possible accept an image topic as input?

FPSychotic, 2020-11-02:
I couldn't find how to do it, would be great if you can make the node accepts an image topic as input, it should be great, in some setups will make you can save have an additional camera ,only for jetson-inference , i.e in a realsense would be possible use the IR right for inference, in the darkness!!!, or use the rgb stream for inference instead of rgb pointclouds. 
this would be useful to accept depth stream as input, as you can republish it and apply AI models over it.

thanks in any case by this amazing repo

dusty-nv, 2020-11-02:
Hi @FPSychotic , each node has an `image_in` topic which you can remap to your image publisher of choice:

https://github.com/dusty-nv/ros_deep_learning#topics--parameters

dusty-nv, 2020-11-02:
If you look at this older documentation, there are examples of using a different image publisher for testing:

https://github.com/dusty-nv/ros_deep_learning/tree/ros1#imagenet-node

FPSychotic, 2020-11-03:
Ahh, alright thanks by that, I will take a look, thank you!

closed 2020-11-03 by FPSychotic