dusty-nv/ros_deep_learning #73: Anyway to integrate my custom model trained ?

FirasAbdennadher, 2021-03-18:
I am looking to integrate my custom model (instance segmentation)

dusty-nv, 2021-03-18:
Hi @FirasAbdennadher, instance segmentation is different than the semantic segmentation that is supported in the underlying jetson-inference library, so you would need to add support for it.  Typically that means the pre/post processing code.  Also you would need your model exported to ONNX and check that TensorRT can load it (you can quickly check that with the trtexec tool found under `/usr/src/tensorrt/bin`)

If you are using PyTorch, you also may want to check out this node which uses torch2trt:  https://github.com/NVIDIA-AI-IOT/ros2_torch_trt

FirasAbdennadher, 2021-03-18:
@dusty-nv Hello , and thank you for this repo and for ur answer , iv trained my custom model based on this official repository in GitHub (matterport/Mask_RCNN:) and tested in my laptop and worked fine,
btw, am newbie in nano Jetson, and am confused how to can i test the model on nano jetson ..
thanks in advance.

MrOCW, 2021-05-07:
Hi @dusty-nv, I have a custom detectnet trained with TLT, how do i interface the TRT related files with ROS?