dusty-nv/ros_deep_learning #76: ros_deep_learning compilation failed

lift-zpz, 2021-05-14:
An error occurred while I was compiling. Here is my error message

jetbot@jetson-4-3:~/workspace/catkin_ws$ catkin_make
Base path: /home/jetbot/workspace/catkin_ws
Source space: /home/jetbot/workspace/catkin_ws/src
Build space: /home/jetbot/workspace/catkin_ws/build
Devel space: /home/jetbot/workspace/catkin_ws/devel
Install space: /home/jetbot/workspace/catkin_ws/install
####
#### Running command: "cmake /home/jetbot/workspace/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/jetbot/workspace/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jetbot/workspace/catkin_ws/install -G Unix Makefiles" in "/home/jetbot/workspace/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jetbot/workspace/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jetbot/workspace/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/jetbot/workspace/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jetbot/workspace/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ros_deep_learning
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ros_deep_learning'
-- ==> add_subdirectory(ros_deep_learning)
detected ROS1 (catkin_make)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetbot/workspace/catkin_ws/build
####
#### Running command: "make -j2 -l2" in "/home/jetbot/workspace/catkin_ws/build"
####
Scanning dependencies of target ros_deep_learning_nodelets
Scanning dependencies of target segnet
[  4%] Building CXX object ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o
[  8%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp: In function ‘bool publish_overlay(uint32_t, uint32_t)’:
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:66:74: error: no matching function for call to ‘segNet::Overlay(imageConverter::PixelType*, uint32_t&, uint32_t&, segNet::FilterMode&)’
  if( !net->Overlay(overlay_cvt->ImageGPU(), width, height, overlay_filter) )
                                                                          ^
In file included from /home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:26:0:
/usr/local/include/jetson-inference/segNet.h:212:7: note: candidate: bool segNet::Overlay(float*, uint32_t, uint32_t, segNet::FilterMode)
  bool Overlay( float* output, uint32_t width, uint32_t height, FilterMode filter=FILTER_LINEAR );
       ^~~~~~~
/usr/local/include/jetson-inference/segNet.h:212:7: note:   no known conversion for argument 1 from ‘imageConverter::PixelType* {aka uchar3*}’ to ‘float*’
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp: In function ‘bool publish_mask_color(uint32_t, uint32_t)’:
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:91:71: error: no matching function for call to ‘segNet::Mask(imageConverter::PixelType*, uint32_t&, uint32_t&, segNet::FilterMode&)’
  if( !net->Mask(mask_color_cvt->ImageGPU(), width, height, mask_filter) )
                                                                       ^
In file included from /home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:26:0:
/usr/local/include/jetson-inference/segNet.h:195:7: note: candidate: bool segNet::Mask(uint8_t*, uint32_t, uint32_t)
  bool Mask( uint8_t* output, uint32_t width, uint32_t height );
       ^~~~
/usr/local/include/jetson-inference/segNet.h:195:7: note:   candidate expects 3 arguments, 4 provided
/usr/local/include/jetson-inference/segNet.h:200:7: note: candidate: bool segNet::Mask(float*, uint32_t, uint32_t, segNet::FilterMode)
  bool Mask( float* output, uint32_t width, uint32_t height, FilterMode filter=FILTER_LINEAR );
       ^~~~
/usr/local/include/jetson-inference/segNet.h:200:7: note:   no known conversion for argument 1 from ‘imageConverter::PixelType* {aka uchar3*}’ to ‘float*’
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp: In function ‘void img_callback(sensor_msgs::ImageConstPtr)’:
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:144:88: error: no matching function for call to ‘segNet::Process(imageConverter::PixelType*, uint32_t, uint32_t)’
  if( !net->Process(input_cvt->ImageGPU(), input_cvt->GetWidth(), input_cvt->GetHeight()) )
                                                                                        ^
In file included from /home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/node_segnet.cpp:26:0:
/usr/local/include/jetson-inference/segNet.h:189:7: note: candidate: bool segNet::Process(float*, uint32_t, uint32_t, const char*)
  bool Process( float* input, uint32_t width, uint32_t height, const char* ignore_class="void" );
       ^~~~~~~
/usr/local/include/jetson-inference/segNet.h:189:7: note:   no known conversion for argument 1 from ‘imageConverter::PixelType* {aka uchar3*}’ to ‘float*’
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp: In member function ‘void ros_deep_learning::ros_imagenet::callback(const ImageConstPtr&)’:
/home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp:88:121: error: no matching function for call to ‘imageNet::Classify(imageConverter::PixelType*, uint32_t, uint32_t, float*)’
             const int img_class = net->Classify(imgCvt->ImageGPU(), imgCvt->GetWidth(), imgCvt->GetHeight(), &confidence);
                                                                                                                         ^
In file included from /home/jetbot/workspace/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp:2:0:
/usr/local/include/jetson-inference/imageNet.h:154:6: note: candidate: int imageNet::Classify(float*, uint32_t, uint32_t, float*)
  int Classify( float* rgba, uint32_t width, uint32_t height, float* confidence=NULL );
      ^~~~~~~~
/usr/local/include/jetson-inference/imageNet.h:154:6: note:   no known conversion for argument 1 from ‘imageConverter::PixelType* {aka uchar3*}’ to ‘float*’
/usr/local/include/jetson-inference/imageNet.h:162:6: note: candidate: int imageNet::Classify(float*)
  int Classify( float* confidence=NULL );
      ^~~~~~~~
/usr/local/include/jetson-inference/imageNet.h:162:6: note:   candidate expects 1 argument, 4 provided
ros_deep_learning/CMakeFiles/segnet.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o] Error 1
CMakeFiles/Makefile2:535: recipe for target 'ros_deep_learning/CMakeFiles/segnet.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/segnet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 13%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o
ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1148: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

dusty-nv, 2021-05-14:
Hi @lift-zpz, I think you need to update your jetson-inference repo.

lift-zpz, 2021-05-14:
> 嗨@ lift-zpz，我想您需要更新您的jetson-推理回购。

> OK, thank you， I will try again

lift-zpz, 2021-05-15:
> 嗨@ lift-zpz，我想您需要更新您的jetson-推理回购。

Hello, after re-cloning, I put the downloaded module under network, and then I executed Vim. gitModules and put the path of updating submodules on the website where the code is stored in China. Finally, I succeeded in downloading and compiling the ROS function package.