dusty-nv/ros_deep_learning #78: Test Classification for single image.

priyaganaboor-hash, 2021-05-30:
Hey, @dusty-nv 

This below command uses the camera, can I also test the working by passing just single image. Similar to how we did in jetson-inference, where we passed the input as path to the image.file.
 
roslaunch ros_deep_learning video_viewer.ros1.launch input:=csi://0 output:=display://0

Which commands I have to change and which launch file to include?

priyaganaboor-hash, 2021-06-01:
hey @dusty-nv ,

can u please help me solve this?

priyaganaboor-hash, 2021-06-01:
If i type like this,

roslaunch ros_deep_learning imagenet.ros1.launch input:=/home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/train_img11.jpg output:=display://0/home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/output1111.jpg

--
[TRT]    device GPU, /home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/resnet50.onnx initialized.
[TRT]    imageNet -- loaded 2 class info entries
[TRT]    imageNet -- /home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/resnet50.onnx initialized.
[ INFO] [1622558727.615601021]: model hash => 6707725573171851291
[ INFO] [1622558727.622362479]: hash string => /home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/resnet50.onnx/home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/labels.txt
[ INFO] [1622558727.627902896]: node namespace => /imagenet
[ INFO] [1622558727.627993729]: class labels => /imagenet/class_labels_6707725573171851291
[ INFO] [1622558727.646194094]: imagenet node initialized, waiting for messages

It is waiting for publisher, what does it mean?

priyaganaboor-hash, 2021-06-10:
Hey @dusty-nv 

can you please help me here??

dusty-nv, 2021-06-10:
@priyaganaboor-hash I think the issue may be that videoSource only publishes the image once, since it is only one image.  What if you just point it to `input:=/home/bm-jetson/catkin_ws/src/ros_deep_learning/PlantClassification/` ?  Then it should publish many images from the directory

priyaganaboor-hash, 2021-06-10:
Thanks @dusty-nv 

I published to topic /video_source/raw, it worked.

once it is classified the plant i want to terminate the execution, I am checking where i can use ros_shutdown, or use required="true" in launch.
which would be better option?

dusty-nv, 2021-06-10:
> I am checking where i can use ros_shutdown, or use required="true" in launch.
Sorry I'm not familiar with these options.  I think `ros_shutdown` only shuts down the node it was run from within.

closed 2021-06-10 by priyaganaboor-hash