dusty-nv/ros_deep_learning #79: provide tutorial video on ROS

sabrabano0, 2021-08-07:
HI 
this is my first experience with  ROS,
i cant found  your tutorial video on this subject similar to before episodes .
i want just use ROSfor this project so i don't need to another references.
thank you

dusty-nv, 2021-08-08:
Hi @sabrabano0, I don't have a video on these ROS nodes.   What I recommend is to go through ROS tutorials to familiarize yourself with ROS first:

http://wiki.ros.org/ROS/Tutorials

Or if you want to use ROS2:  https://docs.ros.org/en/foxy/Tutorials.html

sabrabano0, 2021-08-08:
> Hi @sabrabano0, I don't have a video on these ROS nodes. What I recommend is to go through ROS tutorials to familiarize yourself with ROS first:
> 
> http://wiki.ros.org/ROS/Tutorials
> 
> Or if you want to use ROS2: https://docs.ros.org/en/foxy/Tutorials.html

Thank you for fast reply, you are so active and talent, i noticed you have amazing slides , i wish i have any description with your voice on these slides, have i any luck?

dusty-nv, 2021-08-09:
Hi @sabrabano0, you can find some of my webinar recordings here:

* https://www.youtube.com/watch?v=9gVupqHqJws&t=1534s
* https://www.youtube.com/watch?v=Kag4iPCy2bk&t=10s
* https://www.youtube.com/playlist?list=PL5B692fm6--tMDTpxk_5akMWlyq2n-qmW