dusty-nv/ros_deep_learning #101: Using rostopic echo /detectnet/detections issues

SokPhanith, 2022-02-14:
Hello, I follow your project and then test "roslaunch ros_deep_learning detectnet.ros1.launch input:=csi://0 output:=display://0" working well and using "rostopic echo /detectnet/detections" is working too on a single machine is Jetson nano. 

But When I set up my PC ubuntu as master because I want Jetson Nano detectnet and pub data to another machine as my PC, I start set IP by 

PC :
export ROS_MASTER_URI=http://IP:11311
export ROS_HOSTNAME=IP

Jetson Nano :
export ROS_MASTER_URI=http://IP:11311
export ROS_HOSTNAME=IP_JETSON_NANO

Run "roslaunch ros_deep_learning detectnet.ros1.launch input:=csi://0 output:=display://0" and "rostopic echo /detectnet/detections issues " is working fine on Jetson nano but PC do not working with  "rostopic echo /detectnet/detections", I check "rostopic list" , i saw the same on jetson nano. The Message that I got like below

[image]

dusty-nv, 2022-02-14:
Hi @SokPhanith, I'm not terribly familiar with cross-machine ROS topics, but it looks like your PC needs the [vision_msgs](http://wiki.ros.org/vision_msgs) package installed?

SokPhanith, 2022-02-16:
Thank you it's working now . I forget install vision_msgs on my PC. One more thing I really want flip my camera 180degree. In Jetson inference using --input-flip=rotate-180 but I can't find in ros_deep_learning.

dusty-nv, 2022-02-16:
OK cool, glad you got it working.

See the `flip` parameter of the video_source node:  https://github.com/dusty-nv/ros_deep_learning#video_source-node