dusty-nv/ros_deep_learning #106: subscriber error

sys-met, 2022-05-30:
Hello dusty-nv
I want to receive data with a simple subscriber node, but I am getting the following error
  
 rospy.loginfo("Robot şarjı: %f"%mesaj.bbox.center.x)
AttributeError: 'Detection2DArray' object has no attribute 'bbox'

dusty-nv, 2022-05-31:
Hi @sys-met, see here for the specification of the `Detection2DArray` message:  http://docs.ros.org/en/lunar/api/vision_msgs/html/msg/Detection2DArray.html

I think you probably need to index it to get the `Detection2D` message that you are looking for.

sys-met, 2022-06-05:
Hello dusty-nv
i need a simple example on this

 rospy.loginfo(detections[0].bbox.size_y)
TypeError: 'Detection2DArray' object does not support indexing

sys-met, 2022-06-08:
thanks, it worked

closed 2022-06-08 by sys-met