dusty-nv/ros_deep_learning #107: erro in nodelet_imagenet.cpp

ivory4u, 2022-07-04:
I just get below error.
How can I fix it? 

[  3%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
/home/nvidia/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp: In member function ‘void ros_deep_learning::ros_imagenet::callback(const ImageConstPtr&)’:
/home/nvidia/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp:88:121: error: no matching function for call to ‘imageNet::Classify(imageConverter::PixelType*, uint32_t, uint32_t, float*)’
 g_class = net->Classify(imgCvt->ImageGPU(), imgCvt->GetWidth(), imgCvt->GetHeight(), &confidence);
                                                                                                 ^
In file included from /home/nvidia/catkin_ws/src/ros_deep_learning/src/nodelet_imagenet.cpp:2:0:
/usr/local/include/jetson-inference/imageNet.h:154:6: note: candidate: int imageNet::Classify(float*, uint32_t, uint32_t, float*)
  int Classify( float* rgba, uint32_t width, uint32_t height, float* confidence=NULL );
      ^
/usr/local/include/jetson-inference/imageNet.h:154:6: note:   no known conversion for argument 1 from ‘imageConverter::PixelType* {aka uchar3*}’ to ‘float*’
/usr/local/include/jetson-inference/imageNet.h:162:6: note: candidate: int imageNet::Classify(float*)
  int Classify( float* confidence=NULL );
      ^
/usr/local/include/jetson-inference/imageNet.h:162:6: note:   candidate expects 1 argument, 4 provided
ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/build.make:62: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o' failed
make[2]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o] Error 1
CMakeFiles/Makefile2:3624: recipe for target 'ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all' failed
make[1]: *** [ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2022-07-04:
Hi @ivory4u, I haven't been maintaining this nodelet, so I would just recommend commenting out lines 85-90 of CMakeLists.txt : https://github.com/dusty-nv/ros_deep_learning/blob/a58d6961b06e380f7bd0f2b5379d7f249f7da590/CMakeLists.txt#L85

dusty-nv, 2022-07-04:
Hmm...can you try adding `jetson-utils` to this line:   https://github.com/dusty-nv/ros_deep_learning/blob/a58d6961b06e380f7bd0f2b5379d7f249f7da590/CMakeLists.txt#L79