dusty-nv/ros_deep_learning #116: ros_deep_learning package not found

MiguelOsun, 2023-04-03:
Hi, thank you for all your work.

I am using a Jetson Nano with L4T R32.7.1. so when I run your docker/run.sh -ros=foxy. The problem is that, when I tried to test with roslaunch to launch video_viewer, it said there was no "ros_deep_learning" package. I followed every step in the repo, so I dont really know if it is because I am doing something wrong or it is a problem related to the container itself.

lenhatquang2512, 2023-04-04:
Check my issue :[]( https://github.com/dusty-nv/ros_deep_learning/issues/115).

I used same Nano with same R32.7.1 like you. The container for that version has not been updated yet so there is no pre-installed ros deep learning package. My suggestion is to build from source like I did with the melodic branch version because I think you also used Ubuntu 18.04 for your Nano.

MiguelOsun, 2023-04-04:
Hi! As you said, I am using Ubuntu 18.04 too. I will try building from source with Melodic. Thank you for your help.

dusty-nv, 2023-04-04:
@MiguelOsun I have updated the container images for L4T R32.7.1 to include ros_deep_learning package, see https://github.com/dusty-nv/ros_deep_learning/issues/115#issuecomment-1495956118

MiguelOsun, 2023-04-05:
Hi! Thank you for your help! I am new to Docker Hub, and I have started recently to work with containers. Does this repo use this updated container images, or do i have to update them?

dusty-nv, 2023-04-05:
If you had already tried running the container, you should pull it again to fetch the updated one - for example:   `sudo docker pull dustynv/ros:noetic-pytorch-l4t-r32.7.1`

MiguelOsun, 2023-04-11:
Hi,
as you said, i pulled the image again and i dont have that problem anymore, but when i try to launch any node of the ros_deep_learning package i have this issue: 

[video_output-2] /ros_deep_learning/install/ros_deep_learning/lib/ros_deep_learning/video_output: error while loading shared libraries: libnppicc.so.10: cannot open shared object file: No such file or directory
[video_source-1] /ros_deep_learning/install/ros_deep_learning/lib/ros_deep_learning/video_source: error while loading shared libraries: libnppicc.so.10: cannot open shared object file: No such file or directory

Is this because i have another version of CUDA installed?

dusty-nv, 2023-04-11:
Hi @grafoteka, which version of the container are you running?

grafoteka, 2023-04-11:
> Hi @grafoteka, which version of the container are you running?

Hi @dusty-nv, I'm using the last version that you update a few days ago.  **Galactic-pytorch-l4t-r32.7.1**
https://hub.docker.com/layers/dustynv/ros/galactic-pytorch-l4t-r32.7.1/images/sha256-1fe64222761f7bbc438592d0b22bfa09ec51c92f7f160e71f5bb6ae1cb46369e?context=explore

lenhatquang2512, 2023-04-28:
> Hi, as you said, i pulled the image again and i dont have that problem anymore, but when i try to launch any node of the ros_deep_learning package i have this issue:
>
> [video_output-2] [...]

@MiguelOsun  Which CUDA version are you using? Im using CUDA 10.2

dusty-nv, 2023-05-04:
> im trying to use dustynv/ros:foxy-ros-base-l4t-r32.5.0 container and it not include ros_deep_learning package.

@saarCogni only the `ros:*-pytorch-*` images have ros_deep_learning package in them.   `dustynv/ros:foxy-pytorch-l4t-r32.7.1` has been updated to have ros_deep_learning in it, but I don't have systems still running older versions of JetPack 4.   So you would either need to rebuild the `ros:foxy-pytorch` container for L4T R32.5.0, or install ros_deep_learning into it yourself (which you can do by following this script:  https://github.com/dusty-nv/jetson-containers/blob/master/scripts/ros_deep_learning.sh), or upgrade your JetPack to L4T R32.7.

saarCogni, 2023-05-08:
@dusty-nv 
hi,
when im running the docker `sudo docker run --runtime nvidia -it --rm --network host dustynv/ros:foxy-pytorch-l4t-r32.5.0 bash`
on NVIDIA Jetson Xavier with the following script: [ros_deep_learning.sh](https://github.com/dusty-nv/jetson-containers/blob/master/scripts/ros_deep_learning.sh)
this is what i get:

`root@algokobuki-desktop:/# ./ros_deep_learning.sh 
ROS_DISTRO = foxy
sourcing   /opt/ros/foxy/install/setup.bash
ROS_ROOT   /opt/ros/foxy
ROS_DISTRO foxy
Cloning into 'ros_deep_learning'...
remote: Enumerating objects: 39, done.
remote: Counting objects: 100% (39/39), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 39 (delta 12), reused 18 (delta 7), pack-reused 0
Unpacking objects: 100% (39/39), done.
Building ros_deep_learning package for ROS2 foxy
Starting >>> ros_deep_learning
[Processing: ros_deep_learning]                            
--- stderr: ros_deep_learning                               
detected ROS_DISTRO=ROS_FOXY
CMake Warning at CMakeLists.txt:19 (find_package):
  By not providing "FindVPI.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VPI", but
  CMake did not find one.

  Could not find a package configuration file provided by "VPI" (requested
  version 2.0) with any of the following names:

    VPIConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "VPI" to CMAKE_PREFIX_PATH or set "VPI_DIR"
  to a directory containing one of the above files.  If "VPI" provides a
  separate development package or SDK, be sure it has been installed.

detected ROS2 (ament_cmake)
CMake Warning:
  Manually-specified variables were not used by the project:

    CATKIN_INSTALL_INTO_PREFIX_ROOT

make[2]: *** No rule to make target '/usr/local/cuda/lib64/libnppicc.so', needed by 'video_output'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/video_output.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros_deep_learning [38.9s, exited with code 2]

Summary: 0 packages finished [39.5s]
  1 package failed: ros_deep_learning
  1 package had stderr output: ros_deep_learning`

thanks for the help