dusty-nv/ros_deep_learning #121: CLI for input-codec=mjpeg

Fibo27, 2023-05-07:
Hi Dusty
I am using Jetpack 5.1.1 on a Orin Nano on which I have compiled the ros_deep_learning module. I also have a robot running on a nano and it is streaming video on rtp.
1) On my Orin, I am able to view the streaming video using  "video-viewer --input-codec=mjpeg rtp://@:1234"
2) Also, in jetson inference, I can run any of the modules on the incoming stream like "
"./detectnet.py rtp://@:1234 --input-codec=mjpeg"
However when i try to use any of the launch files (I am running Foxy on both the Orin and the Nano on the Robot)
"ros2 launch ros_deep_learning detectnet.ros2.launch input:=rtp://@:1234 output:=display://0" I would expect to get an error which i get since the default input-codec is h264. Although you mention (as does the code) that the input codec can be changed, I get errors when i try either of the two options mentioned above i.e. using 
1) ros2 launch ros_deep_learning detectnet.ros2.launch input-codec:=mjpeg input:=rtp://@:1234 output:=display://0
2) ros2 launch ros_deep_learning detectnet.ros2.launch --input-codec=mjpeg input:=rtp://@:1234 output:=display://0

Can you please help me the syntax for CLI? Thanks

dusty-nv, 2023-05-08:
@Fibo27 can you try adding this line of code under here:

https://github.com/dusty-nv/ros_deep_learning/blob/84485b93eb69aa06ae403ffdacec9a7d5519f49d/launch/video_source.ros2.launch#L14

``` python
<param name="codec" value="$(var input_codec)"/>
```

Unfortunately it appears that I missed that one for mapping `input_codec`, sorry about that.

Fibo27, 2023-05-08:
Thanks Dusty - no need to apologize - you have done an incredible job with these repos.
The suggested change works like a charm. The syntax to be used is input_codec=mjpeg

On another note I had the issue which has been posted in 115 - it relates to ros_deep_learning not being found after colcon build. I had the same issue - since I am using Foxy - i removed lines 22 through 44, 85 to 92 and 111 in cmakelists.txt and ros1 related xml files from the folder - it solved the problem!

dusty-nv, 2023-05-08:
Ah okay!, great thanks 👍   glad you were able to get both those things working :)

closed 2023-05-08 by Fibo27