dusty-nv/ros_deep_learning #122: Jetson inference

vijayreddysamula, 2023-05-20:
Hi, Dusty,
             I am working Jetson Orin AGX developer kit.
               I encountered a small problem after building the ros image. The container doesn't have jetson inference and jetson utils. But I tried cloning the jetson inference and it always shows **webrtc** error.  I don't want to build a jetson inference again, so I wrote the docker file with the jetson inference image as the base and installed ros2. But there I can see the ros2 topic list but not able to subscribe.

dusty-nv, 2023-05-22:
Hi @vijayreddysamula, which version of jetpack-l4t are you running?   (you can check this with `cat /etc/nv_tegra_release`)

You can build the ros:pytorch container (which has ros_deep_learning installed and uses jetson-inference as base) with [jetson-containers](https://github.com/dusty-nv/jetson-containers) `scripts/docker_build_ros.sh --distro=humble --with-pytorch`

vijayreddysamula, 2023-05-22:
Hi Dusty,
        I am using L4t r34.1.1. I tried jetson-containers with the above script using foxy, but i got following error **pull access denied for jetpack, repository does not exist or may require 'docker login': denied: requested access to the resource is denied**

dusty-nv, 2023-05-22:
Hi @vijayreddysamula, IIRC the l4t-jetpack base container for L4T R34.1.1 wasn't released - L4T R34.1.1 was a Developer Preview release and you should upgrade it (i.e. to the latest - currently JetPack 5.1.1 / L4T R35.3.1)

jiefengsun, 2023-05-24:
> Hi @vijayreddysamula, which version of jetpack-l4t are you running? (you can check this with `cat /etc/nv_tegra_release`)
>
> You can build the ros:pytorch container (which has ros_deep_learning [...]

@dusty-nv This means we can directly do inference in the same node by calling imagenet API from jetson-interface. Is this a solution to avoid nodelet?

dusty-nv, 2023-05-24:
@jiefengsun in ROS2 it doesn't use nodelets, but yes.  Isaac ROS also has zero-copy transport between nodes (NITROS)  https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros