dusty-nv/ros_deep_learning #127: bbox information via the /detectnet/detections topic using a subscibrer

kleanbotmk2, 2023-07-31:
Hello,

I wanted to know how I could retrieve bbox information via the /detectnet/detections topic using a subscibrer. I use a node in python under ros2 foxy. Otherwise I wanted to thank you because the package works very well!

Best regards,

Sacha

dusty-nv, 2023-08-01:
Hi @kleanbotmk2, the message type of that topic is `vision_msgs/Detection2DArray`:

http://docs.ros.org/en/api/vision_msgs/html/msg/Detection2DArray.html

dusty-nv, 2023-08-01:
I don't have an example of doing it with rospy, but subscribing to topics and dealing with standard messages is a typical ROS thing.   It's not a custom message type.   You can see how I construct the messages in the C++ code for the detectnet node though:

https://github.com/dusty-nv/ros_deep_learning/blob/master/src/node_detectnet.cpp

kleanbotmk2, 2023-08-28:
okay thanks you it's working

closed 2023-08-28 by kleanbotmk2