dusty-nv/ros_deep_learning #139: gst-launch-1.0 preview is just a black video - no error messages

andanders, 2024-07-15:
I am unable to use my camera inside the docker container. Testing with `gst-launch-1.0 v4l2src ! videoconvert ! autovideosink`i get a preview window with just a black screen.
The same command runs just fine outside the container. So I know the camera works with v4l2src.

To start the container I run
```bash
$ docker/run.sh
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R35.4.1
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r35.4.1
DATA_VOLUME:      --volume /home/nvidia/jetson-ros-inference/data:/jetson-inference/data --volume /home/nvidia/jetson-ros-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/nvidia/jetson-ros-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/nvidia/jetson-ros-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/nvidia/jetson-ros-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/nvidia/jetson-ros-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
V4L2_DEVICES:     --device /dev/video0 
DISPLAY_DEVICE:   -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix 

root@nvidia-desktop:/jetson-inference# 
``` 

To start the video preview;
```bash
$ gst-launch-1.0 v4l2src ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
$ ^C
handling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:04.136207413
Setting pipeline to NULL ...
Freeing pipeline ...

``` 

## System/software info
Linux nvidia-desktop 5.10.120-tegra aarch64 
Docker version 27.0.3, build 7d4bcd8

Repo pulled with `git clone --recursive --depth=1 https://github.com/dusty-nv/jetson-inference`
I'm on the master branch, commit e8361ae7f5f3651c4ff46295b193291a93d52735.

andanders, 2024-07-15:
Another test, maybe it is relevant.
`test-video.py` fails with the same timeout message.

`/jetson-inference/build/aarch64/bin# python3 test-video.py` 

I want to hightlight from the below output:

> failed to find/open file /proc/device-tree/model

> [gstreamer] gstEncoder -- hardware encoder not detected, reverting to CPU encoder

```bash
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0230 30-0043
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0230\ 30-0043", v4l2.device.bus_info=(string)platform:tegra-capture-vi:1, v4l2.device.version=(uint)330360, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 10 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)30/1;
[gstreamer] [1] video/x-raw, format=(string)UYVY, width=(int)1280, height=(int)960, framerate=(fraction)34/1;
[gstreamer] [2] video/x-raw, format=(string)UYVY, width=(int)1280, height=(int)720, framerate=(fraction)45/1;
[gstreamer] [3] video/x-raw, format=(string)UYVY, width=(int)960, height=(int)540, framerate=(fraction){ 58/1, 30/1 };
[gstreamer] [4] video/x-raw, format=(string)UYVY, width=(int)640, height=(int)480, framerate=(fraction){ 60/1, 45/1 };
[gstreamer] [5] video/x-raw, format=(string)NV16, width=(int)1920, height=(int)1080, framerate=(fraction)30/1;
[gstreamer] [6] video/x-raw, format=(string)NV16, width=(int)1280, height=(int)960, framerate=(fraction)34/1;
[gstreamer] [7] video/x-raw, format=(string)NV16, width=(int)1280, height=(int)720, framerate=(fraction)45/1;
[gstreamer] [8] video/x-raw, format=(string)NV16, width=(int)960, height=(int)540, framerate=(fraction){ 58/1, 30/1 };
[gstreamer] [9] video/x-raw, format=(string)NV16, width=(int)640, height=(int)480, framerate=(fraction){ 60/1, 45/1 };
[gstreamer] gstCamera -- selected device profile:  codec=raw format=uyvy width=1920 height=1080 framerate=30
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! video/x-raw, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=30/1 ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- codecType:  cpu
  -- width:      1920
  -- height:     1080
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
------------------------------------------------
failed to find/open file /proc/device-tree/model
[gstreamer] gstEncoder -- detected board 'NVIDIA Orin Nano/NX with Syslogic BRMA3N-11x Carrier'
[gstreamer] gstEncoder -- hardware encoder not detected, reverting to CPU encoder
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! x264enc name=encoder bitrate=2500 speed-preset=ultrafast tune=zerolatency ! video/x-h264 ! h264parse ! qtmux ! filesink location=images/test/test_video.mp4 
[video]  created gstEncoder from file:///jetson-inference/build/aarch64/bin/images/test/test_video.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///jetson-inference/build/aarch64/bin/images/test/test_video.mp4
     - protocol:  file
     - location:  images/test/test_video.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      H264
  -- codecType:  cpu
  -- frameRate:  30
  -- bitRate:    2500000
  -- numBuffers: 4
  -- zeroCopy:   true
[... 43 lines cut ...]
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[... cut ...]
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer

```

dusty-nv, 2024-07-15:
@andanders are you able to see the video stream with `nvgstcapture-1.0` tool?   Normally I try this first with MIPI CSI camera.  This model has some custom formats in YUYV/NV16 format that I have not tried before.  Also with `video-viewer` you want to use `csi://0` for MIPI CSI camera (`nvarguscamerasrc` in GStreamer pipeline or if using gst-launch), not V4L2 `/dev/video0` - the V4L2 driver does not use ISP.

andanders, 2024-07-17:
For the sake of clarity; the camera and computer combo that I'm using is [e-con sturdecam 20](https://www.syslogic.ai/accessories/e-con-systems-sturdecam20-gmsl2-hdr-camera), and [Syslogic RS A4NA](https://www.syslogic.ai/jetson-orin-nano/rugged-computer-rpc-rs-a4na). They are connected with GMSL and a built-in deserializer. So I don't know if it is strictly MIPI or CSI.

<details>
<summary>system detects the camera as ...</summary>

```bash
~$ sudo v4l2-ctl --list-devices
NVIDIA Tegra Video Input Device (platform:tegra-camrtc-ca):
	/dev/media0

vi-output, ar0230 30-0043 (platform:tegra-capture-vi:1):
	/dev/video0
```

</details>

Anyhow. 

**nvgstcapture-1.0**
Directly on host machine

The following command opens a black window (no video), and a "no cameras available"
```bash
~$ nvgstcapture-1.0
Encoder null, cannot set bitrate!
Encoder Profile = High
Codec not supported. Falling back to opensrc H264 encoder
Supported resolutions in case of ARGUS Camera
  (2) : 640x480
  (3) : 1280x720
  (4) : 1920x1080
  (5) : 2104x1560
  (6) : 2592x1944
  (7) : 2616x1472
  (8) : 3840x2160
  (9) : 3896x2192
  (10): 4208x3120
  (11): 5632x3168
  (12): 5632x4224

Runtime ARGUS Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Set sensor orientation:
      so:<val>
          (0): none
          (1): Rotate counter-clockwise 90 degrees
          (2): Rotate 180 degrees
          (3): Rotate clockwise 90 degrees
  Get sensor orientation:
      gso
  Set sensor mode:
      smo:<val> e.g., smo:1
  Get sensor mode:
      gsmo
  Set Whitebalance Mode:
      wb:<val>
          (0): off
          (1): auto
          (2): incandescent
          (3): fluorescent
          (4): warm-fluorescent
          (5): daylight
[... 170 lines cut ...]
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer message stream-start ==> pipeline0
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:751 No cameras available
[gstreamer] gstCamera -- end of stream (EOS)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer message warning ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstreamer pipeline0 recieved EOS signal...
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
```
</details>

andanders, 2024-07-17:
I almost forgot to test with nvarguscamerasrc

```bash
~$ gst-launch-1.0 nvarguscamerasrc ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...

(gst-launch-1.0:6516): GLib-CRITICAL **: 14:43:07.565: g_thread_join: assertion 'thread' failed

(gst-launch-1.0:6516): GLib-CRITICAL **: 14:43:07.565: g_thread_join: assertion 'thread' failed

(gst-launch-1.0:6516): GLib-CRITICAL **: 14:43:07.565: g_queue_is_empty: assertion 'queue != NULL' failed

(gst-launch-1.0:6516): GLib-CRITICAL **: 14:43:07.565: g_queue_free: assertion 'queue != NULL' failed

(gst-launch-1.0:6516): GLib-CRITICAL **: 14:43:07.565: g_queue_free: assertion 'queue != NULL' failed
Freeing pipeline ...
```

dusty-nv, 2024-07-17:
@andanders if nvarguscamerasrc and nvgstcapture aren't working, then I would consult with your camera manufacturer to make sure you have the right drivers or version of JetPack installed to support it.

andanders, 2024-08-20:
If I modify docker/run.sh by adding --privileged, 
I can run `/build/aarch64/bin/video-viewer /dev/video0` and get a preview inside the container.

In a response from E-con systems:
> Our Camera STURDeCAM20 has an on-board ISP and does not use NVIDIA ISP. The cameras which use NVIDIA ISP can only use nvgstcapture and nvarguscamerasrc. 

Though our cameras does not use NVIDIA ISP, it can be streamed through NVIDIA ISP by using nvv4l2camerasrc.

To stream STURDeCAM20 via NVIDIA ISP  use the below command:

$ gst-launch-1.0 nvv4l2camerasrc device=/dev/video<video node> ! 'video/x-raw(memory:NVMM), format=(string)UYVY, width=(int)<width>, height=(int)<height>' ! nvvidconv !  'video/x-raw(memory:NVMM), format=(string)I420, width=(int)<width>, height=(int)<height>' ! nv3dsink sync=false -v

I'm not sure what I'm supposed to do with that information.
If running the container as privileged works, then thats my lot.

But this just gives me other problems down the road, because the ros_deep_learning examples fail.