dusty-nv/ros_deep_learning #144: Pre-built container for ROS Noetic with L4T 35.4.1

amir-zviran, 2024-12-10:
Hi Dustin,
I'm using Jetson Xavier NX with Jetpack 5.1.2 (L4T 35.4.1).
I followed your instructions on the README file to pull the pre-built container for ROS Noetic, but it seemed like there was no such container named dustynv/ros:noetic-pytorch-l4t-r35.4.1.
Looking at jetson-containers, I saw there was one named dustynv/ros:noetic-pytorch-l4t-r35.3.1.
How can I build a similar one for L4T 35.4.1?
Thanks!
Amir

amir-zviran, 2025-01-22:
@dusty-nv ?

dusty-nv, 2025-01-22:
Hi @amir-zviran, I would try `jetson-containers build --name=my_base ros:noetic jetson-inference` to build that base.   That image was effectively ROS + pytorch + jetson-inference